horizon-eda/horizon #682: Enhance Schematic Assign view

tkornack, 2022-08-01:
It would be great if the Assign view were larger by default. Here's what mine looks like:

[image]

It seems to be necessary to first expand the window and then expand the MPN column before any useful selection can be made. Can the MPN column be wider by default? 

It might be better if the MPN were not populated with the existing part number so that you can see all the possible choices.

And finally most of the time I'm tweaking resistors and capacitors so it would be especially great if the parametric search showed up here as well.

carrotIndustries, 2022-08-01:
That dialog is more or less just a remnant from before when we had the part browser window. 

The proper way to assign parts is selecting the to-be-assigned symbols, select the right part in the part browser and click "assign" in there,

tkornack, 2022-08-01:
That is ideal. Nevermind!

closed 2022-08-01 by tkornack

carrotIndustries, 2022-08-01:
I think this still leaves the question if there's a a need to keep the "assign part" tool as it's mostly been replaced by the part browser. The only thing the part browser can't do is clearing the assigned part, though that could be put into its own tool. Also, the part browser isn't limited to parts of the matching entity.

tkornack, 2022-08-01:
Can the part browser grow an entity filter, and the assign part command would just open the part browser and specify the entity?

carrotIndustries, 2022-08-13:
> Can the part browser grow an entity filter, and the assign part command would just open the part browser and specify the entity?

That's more or less what's in the recent commits on `master` now.