horizon-eda/horizon #683: Too much exception thrown due to unimplemented functions. after applying Kicad router 6.0.4.

joshuawins, 2022-08-16:
I don't want to complain about the features of version that are not released yet, but the new router seems so fantastic that I want to use it as early as possible.
While I was testing the router, program crashed without warning or notification, since it throwed unhandled exception. I guess you already are working on it. The frequent crash was due to unimplemented functions in the file pns_horizon_iface.cpp
HoleClearance(const PNS::ITEM *aA, const PNS::ITEM *aB)
HoleToHoleClearance(const PNS::ITEM *aA, const PNS::ITEM *aB)
...
I hope you can implement soon.
I have switched to released version now.

carrotIndustries, 2022-08-16:
I haven't implemented them yet since I've never seen them called. Can you provide a minimum working example that allows me to reproduce this issue?

joshuawins, 2022-08-17:
[crashtest.zip](https://github.com/horizon-eda/horizon/files/9355548/crashtest.zip)
I have made a crash project. I have put silk screen note on it.

joshuawins, 2022-08-23:
Your fix works great! Now I don't see sudden crash anymore.
Thank you, Lukas, for your great job!

commit e9e8346d (2022-08-21): router: don't crash in Dbg() is NULL

should fix #683
commit 22bdf237 (2022-08-24): router: don't crash in Dbg() is NULL

should fix #683
commit f1de91fe (2022-08-27): router: don't crash in Dbg() is NULL

should fix #683
closed 2022-08-23 by joshuawins