horizon-eda/horizon #685: Error in automatic ZIP generation, oshpark fails to parse

timk2022, 2022-08-28:
**Version**: (2.3.1 Iceblink Commit 435b4e7 2022-05-29 20:00:44 +0200 version 2.3.1)
**Issue**:
I am on the windows version of horizon. I have been trying to upload the automatically generated zip to oshpark for my project but it has been failing. **The fix is to manually zip the generated files** (I have run diffs on them, they are otherwise all the same). It seems that there is an issue in the zipping run by horizon. I have attached the two versions, the auto generated one seems to be about 800 bytes larger. I have not gone into specifics for debugging this yet.
[out-self-zip.zip](https://github.com/horizon-eda/horizon/files/9440445/out-self-zip.zip)
[out-auto-zip.zip](https://github.com/horizon-eda/horizon/files/9440446/out-auto-zip.zip)

carrotIndustries, 2022-08-28:
Interesting, I've wrote them an email so that we get to know the exact error that causes the upload to fail.

carrotIndustries, 2022-08-29:
Well, that was a strange one. We don't set the actual time and date in the generated zip to keep things more reproducible. The fabrication output should be a function of the board (and if needed the version of Horizon EDA) and not depend on the phase of the moon.

commit 7a0db52e (2022-08-29): TreeWriter: set mtime/ctime to not too far in the past date

closes #685

apparently, OSH Park doesn't like files created in the 80s

changelog: Bugfixes/Board Editor: Fabrication output: set modification
time for improved compatibility with board manufacturers
closed 2022-08-29 by carrotIndustries