horizon-eda/horizon #689: gl error 1281 in src/canvas/triangle_renderer.cpp:120

b99, 2022-09-16:
My OS is Windows7 X64.
When Click on any of the Schematic / Board / Part Browser / Project Pool , in a while show the error:

gl error 1281 in src/canvas/triangle_renderer.cpp:120
Program will abort

The releases v2.3.1, and the last git version are same error.

carrotIndustries, 2022-09-16:
That might be your GPU/drivers not supporting OpenGL 3.

Also, Windows 7 isn't supported by Horizon EDA as it's EOL and libraries are phasing out support for it as well: https://github.com/msys2/MINGW-packages/issues/10088 I should probably add this to the documentation.

carrotIndustries, 2022-10-15:
> I should probably add this to the documentation.

Done: https://horizon-eda.readthedocs.io/en/latest/installation.html#windows

closed 2022-10-15 by carrotIndustries