horizon-eda/horizon #691: 3D Flying parts with Jupiter

bob-u, 2022-10-16:
Congratulations on the new release.  I tried newly the Jupiter version with example project x-transmitter and I got misplaced 3D parts in 3D preview:

[image]
[image]

carrotIndustries, 2022-10-16:
There were no changes to the 3D view from 2.3 to 2.4. To me this looks more like a driver issue since I've seen similar things happen on AMD GPUs on Windows.

bob-u, 2022-10-16:
Yes, I'm on Windows and AMD Ryzen 5500.  I'll upgrade the drivers.

bob-u, 2022-10-16:
I updated to the latest AMD driver dated 10/3/2022 (my old driver was from 2021) and reset all settings which resolved the issue.  All 3D components have returned to earth.

carrotIndustries, 2022-10-16:
> I updated to the latest AMD driver dated 10/3/2022 (my old driver was from 2021) and reset all settings which resolved the issue. All 3D components have returned to Earth.

Ok, I guess we can close this issue then?

closed 2022-10-16 by carrotIndustries

bob-u, 2022-10-17:
Yes, it was a driver issue.