horizon-eda/horizon #693: ld.gold: Changing linker on commandline

maikwoehl, 2022-10-29:
Is this variable ment to be overwritten from commandline - like `WITH_SPNAV`? Does it missing a `?` here?
https://github.com/horizon-eda/horizon/blob/750111e829cd6f3563ef8040c224ce147808c27d/Makefile#L958

As a maintainer for the package in Solus I've now included a patch to replace `-fuse-ld=gold` with `-fuse-ld=bfd` into my package build description and not using clang anymore.

https://dev.getsol.us/D13683

carrotIndustries, 2022-10-29:
For me, `make GOLD=-fuse-ld=lld ` does work.

maikwoehl, 2022-10-29:
Okay, I'll give it a try - maybe I've done something wrong or our build environment strips away that variables. Wouldn't be surprising. 

Thanks for your quick reply.

closed 2022-11-23 by maikwoehl