horizon-eda/horizon #696: Feature request: assign base part to existing part

tkornack, 2022-11-07:
I often create a part such as a diode that's part of a series without a base part. Later, if I need a second member of that series, I'll have to create a base part and re-create the old part derived from the base part. Would it be worth the effort to develop the following features?

- assign a base part to an existing part

and/or:

- create a base part based on an existing part, and have the existing part inherit from its new parent

and while we're at it:

- make a part independent of its base part.

tkornack, 2022-11-07:
Also, in my existing workflow, when I replace the existing (baseless) part file with a new one (with a base part), the UI kindly warns about replacing the existing part file, but then the system start throwing a lot of errors until it's restarted. Not sure what should be done about that sort of behavior, but hopefully a feature like this would eliminate the issue.

carrotIndustries, 2022-11-09:
I'll add a button in the part editor to allow changing the base part to another part of the same entity.

>    create a base part based on an existing part, and have the existing part inherit from its new parent

Once the part editor supports changing base part, this will be next.

> make a part independent of its base part.

This will also be covered by the change base part feature.

Implementing all of this will be a bit tedious since the part editor has been built with the assumption that a part's base part and there being a base part at all won't change.

tkornack, 2022-11-09:
Thanks for considering this!

carrotIndustries, 2022-11-13:
See the linked branch for an implementation of this. I'm not quite sure about the workflow for inserting a base part. Right now, the user enters the filename of the new base part and it's saved and the current part is set to use it. Editing the new base part requires clicking on it and going back to the pool manager to open it. Should we open it right away? Any other suggestions for how the workflow could look like?

tkornack, 2022-11-14:
This is precisely what I had in mind and works perfectly in my testing. I do think a newly created base part should be opened right away; that's quite natural.

Thank you!

carrotIndustries, 2022-11-19:
> I do think a newly created base part should be opened right away; that's quite natural.

dbacba89fc364aadb46b115b9e35af39585865e9 adds this

commit b58e542c (2022-11-19): part editor: add creating base part

changelog: Enhancements/Part Editor: support creating base part from current part

closes #696
closed 2022-11-19 by carrotIndustries