horizon-eda/horizon #697: Feature request: Part replacement migration

tkornack, 2022-11-07:
As part availability changes, it would be a handy feature to let an old part point to its drop-in replacement part, and have a method for updating old parts with their replacements. I find it particularly cumbersome to assign new parts for every unavailable passive component on every design.

carrotIndustries, 2022-11-07:
> I find it particularly cumbersome to assign new parts for every unavailable passive component on every design.

For passive parts,try the "similar parts" tab in the BOM export dialog.

tkornack, 2022-11-08:
> 

> For passive parts,try the "similar parts" tab in the BOM export dialog.

I like the Similar Parts section. Perhaps an option there to make the parts selected be permanent?

carrotIndustries, 2022-11-08:
> Perhaps an option there to make the parts selected be permanent?

I guess that'd be confusing, how about an import/export feature instead?

tkornack, 2022-11-08:
> how about an import/export feature

Not sure I can picture how it would work. Would you export the BOM as normal, but then import it back in to make the assignments permanent?

carrotIndustries, 2022-11-08:
> Not sure I can picture how it would work.

Rather than redoing the assignments manually, they can be imported from a file that has been exported before.