horizon-eda/horizon #699: Feature request: toggle visibility of individual pictures

bsilver8192, 2022-11-21:
I would like some way to toggle the visibility of individual pictures in the package editor. Maybe leave a visible border so it's still possible to find them? Or an option to temporarily hide all pictures except a specific one, like the Isolate command in solidworks?

Context: I'm doing a footprint for both https://www.infineon.com/cms/en/product/packages/PG-HSOG/PG-HSOG-8-1/ and https://www.infineon.com/cms/en/product/packages/PG-HSOF/PG-HSOF-8-1/. The land patterns are very similar, and other documentation confirms they are compatible, but there's no single footprint for both of them. I've found adding both pictures helpful, but just changing one to .1 opacity (the current minimum) makes it hard to tell which lines are from which.

carrotIndustries, 2022-11-21:
How about adding a name field to pictures? The name will then also show up in the clarify selection menu.

Another option is to allow a minimum opacity of 0, but show a outline in that case.

bsilver8192, 2022-11-21:
> How about adding a name field to pictures? The name will then also show up in the clarify selection menu.

That would also be nice, but I don't think it solves the problem. I would like to get several pictures positioned and then switch between them. I don't see a way to toggle visibility of individual pictures currently.

> Another option is to allow a minimum opacity of 0, but show an outline in that case.

That sounds great.

carrotIndustries, 2022-11-26:
See the linked branch for an implementation of this.