horizon-eda/horizon #705: Support for InteractiveHtmlBom

trembel, 2023-01-22:
I really like to use [InteractiveHtmlBom](https://github.com/openscopeproject/InteractiveHtmlBom) when soldering my PCBs, so would be nice to have support for it also from Horizon. The protocol is described here: https://github.com/openscopeproject/InteractiveHtmlBom/wiki/Compatibility

frmdstryr, 2023-10-12:
Not sure if you saw it, but Horizon already has an interactive "Parts list" that you can open from the board editor. 

When open, you select a part/bom item in the list it highlights where components go in both the editor and 3d view. 

[image]

frmdstryr, 2023-10-12:
I don't see anything in the docs so perhaps that could be added.