horizon-eda/horizon #706: Feature request: Custom values in BOM export

drakedevel, 2023-02-05:
Rather than creating separate parts for every value resistor I need, I've been using the "base 0603 resistor" part, and setting the value in the schematic. However, when I use the BOM Export feature, the Value column is populated with the default value (the string "base 0603 resistor") instead of the custom value that shows up on the schematic.

Given that I'm using the BOM to buy parts, it would be nice to have the custom value somewhere. It seems like it should probably just override the Value column, but I would be fine with a new Custom Value column as well.

The one complication is that the rows would need to be split by that value, since right now all of the "base 0603 resistor"s are grouped in one row. I don't know if there are other uses for custom values that that would break (doesn't seem likely).

carrotIndustries, 2023-02-10:
> Rather than creating separate parts for every value resistor I need, I've been using the "base 0603 resistor" part, and setting the value in the schematic

That's not what the custom value is for. The custom value is for adding additional information from the parmetric fields to the schematic.

> The one complication is that the rows would need to be split by that value, since right now all of the "base 0603 resistor"s are grouped in one row. 

That's exactly the reason why we have parts in the first place - not having to reconcile components into BOM lines based on manual input.

drakedevel, 2023-02-10:
> The custom value is for adding additional information from the parmetric fields to the schematic.

Is the idea that if you have a 1kOhm 1% resistor part in the schematic, the displayed value will just be "1 kΩ", and you can use the custom value field to change it to "1 kΩ 1%" if that's relevant? Or am I misunderstanding?

> That's exactly the reason why we have parts in the first place - not having to reconcile components into BOM lines based on manual input.

Is there a recommended alternative way to specify generic passives? I didn't find anything like that in the pool as it is today. Sometimes I do want to specify a specific passive because there's some particular properties I need (DC bias on a cap, ESR on an inductor, etc.), but most of the time I just want whatever's the cheapest part on DigiKey/Newark for the quantity I'm ordering.

I could go find the cheapest MPN on DigiKey at the time I'm doing the schematic and make a part for it. But that creates new problems:
- I like to re-use blocks for things like power supplies, and I don't really want a BOM with 5 different manufacturers' 0.1uF X7R capacitors if I re-use blocks that I made at different times.
- I'd rather not come back to a board in six months and find I can't order parts for another copy because the specific 5% 1kOhm pullup resistor I specified is discontinued / out of stock.

For a little bit of background, I'm a former EAGLE user, and years ago I wrote some code to parse the EAGLE schematic, solve for the cheapest parts that meet the constraints I specified in the schematic part attributes, and spit out a CSV that I can paste into a Digi-Key cart and hit order on. If I needed a specific passive I added a `MPN` attribute. I definitely don't want Horizon to copy EAGLE here, but is there a way you see that's compatible with your vision for Horizon that would enable a workflow like that?

One idea could be to create a bunch of parts like "1kΩ 1% 0603 resistor" with appropriate parametric fields? It would be easy enough to automate, but I can't shake the feeling that it feels a little bit odd to have hundreds of parts that only differ in a programmatic way.

tkornack, 2023-02-10:
> I like to re-use blocks for things like power supplies, and I don't really want a BOM with 5 different manufacturers' 0.1uF X7R capacitors if I re-use blocks that I made at different times.

This is a real problem, but the solution isn't to make a generic passive. I think there ought to simply be a better workflow in the Similar Parts area. I was advocating for some kind of migration path for obsolete parts. Here's a similar issue https://github.com/horizon-eda/horizon/issues/697 .

drakedevel, 2023-02-11:
> This is a real problem, but the solution isn't to make a generic passive.

Ironically, the only 0603 0.1uF X7R capacitor in the pool is a generic passive! It has the non-existent manufacturer "CPL", and the model number matches the scheme in the now-abandoned Octopart [Common Parts Library](https://octopart.com/blog/archives/2015/07/introducing-cpl-part-numbers) of (generic) passives.

> I think there ought to simply be a better workflow in the Similar Parts area.

I don't see an obvious way that would work for me. In my old workflow, the constraints on passives are encoded in the schematic, or a specific MPN if the constraints are infeasible to solve via parametric search (e.g. Digi-Key has no data on DC bias performance).

In Horizon, if I'm always selecting a specific MPN, those constraints aren't encoded anywhere. No automation can tell it whether I picked a MPN because it was the cheapest one on Digi-Key or because it's part of a network-analyzer-tuned RF filter. If I have to go through every passive in the schematic by hand to decide whether it's safe to dedupe that is a big step backwards for me (and a source of avoidable human error).

To be clear, I'm not saying that everyone should have to do it this way! I'm just looking for a way to avoid wasting time thinking about non-critical passives, and I suspect others would find it useful, too. (But I guess that's a different feature request.)

> Here's a similar issue #697 .

The Similar Parts workflow works somewhat better for obsolete parts, since it's only needed when a part becomes unavailable, instead of every time you re-use a block. But it has the same basic problem in that the constraints aren't encoded anywhere, so Horizon can't know whether a simple parametric search is good enough, or what parameters matter.

tkornack, 2023-02-11:
There are many possible ways to specify genericness. What would solve the problem is to specify a replacement rule or migration path for every obsolete or unavailable MPN. It could be list that just specifies UUID A is replaced by UUID B, so long as the parts share the same entity and package. Then you get to choose precisely which things get to change and in which ways.  

For your RF tuned circuit, it seems that you're always going to have a problem if you use a generic part that is not sufficiently specified or precise.

RX14, 2023-02-21:
> In Horizon, if I'm always selecting a specific MPN, those constraints aren't encoded anywhere. No automation can tell it whether I picked a MPN because it was the cheapest one on Digi-Key or because [...]

This is what the custom value field is for! You select the specific part you chose for this BOM but set the value to describe why you chose the part. Changing out parts doesn't affect anything in the schematic, just the BOM (or board if there's a different footprint). The difficult part is when that part is not already in an existing pool. The fix here which is most in line with how horizon is supposed to be used is to autogenerate passives for all major suppliers.

This has been done for a good number of the common suppliers such as Murata, TDK, yageo, Panasonic and Samsung. But the libraries don't have more specialist product lines, and for example panasonic only has resistors, not capacitors. The core issue here is maintenance of the pools by horizon contributors, a task which is tedious and has been left by the wayside since adoption of horizon remains fairly small—in part because of these pool issues.

An automated tool could with some effort automatically parse the custom value, find the cheapest part with that description (from approved suppliers with horizon pools) and replace the part in the schematic, just like your eagle script. But this is unlikely to be built into horizon, since schematic comments are mostly designed to be read by humans, not scripts, and most users will simply stick to one passive supplier for resistors and capacitors, because the main costs of a project is not in passives but other items.