horizon-eda/horizon #707: Feature request: no paste openings for unpopulated parts

tkornack, 2023-02-10:
When a part is not populated, it is often desirable to omit the solder paste from the stencil. 

I believe this behavior is found in other ECAD systems.

It would be particularly harmonious to implement this with https://github.com/horizon-eda/horizon/issues/603 to make the 3D and STEP files also reflect the unpopulated components.

carrotIndustries, 2023-02-11:
> I believe this behavior is found in other ECAD systems.

Which ones? As per my understanding, assembly variants (the more generalized form of "do not populate") should only affect the BOM and nothing else.

I could see this feature complicating the addition of assembly variants in the future.

Also, most if not all commercially-made boards I've seen so far had solder on unpopulated components.

tkornack, 2023-02-11:
Eagle and Fusion did it. It's a bit easier to add an unpopulated part since you don't have to remove the solder first, so that was the motivation.

On the other hand, pads without solder will oxidize a bit over time, so it's not always the correct choice.

tkornack, 2023-02-11:
You can close this if you don't have an interest in implementing this.