horizon-eda/horizon #715: Library packages conflicts when building on linux

erusyd, 2023-04-03:
I want to use latest horizon eda on my Linux Mint 21.1, but when I install dependencies following instructions on https://horizon-eda.readthedocs.io/en/latest/build-linux.html, I get following error:
```
The following packages have unmet dependencies:
 kicad : Conflicts: liboce-foundation11 but 0.18.3-2 is to be installed
         Conflicts: liboce-modeling11 but 0.18.3-2 is to be installed
         Conflicts: liboce-ocaf11 but 0.18.3-2 is to be installed
         Conflicts: liboce-visualization11 but 0.18.3-2 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```
So is it possible to do a static build on another Linux Mint machine?

carrotIndustries, 2023-04-03:
I can't reproduce this on a fresh ubuntu 22.04 VM with kicad installed from the repos. Where did you install kicad from?

erusyd, 2023-04-03:
I installed KiCad from this ppa:
https://launchpad.net/~kicad/+archive/ubuntu/kicad-7.0-releases

erusyd, 2023-04-03:
I think it is following libraries conflict with liboce-xxx:
```
libocct-data-exchange-7.5
  libocct-foundation-7.5 libocct-modeling-algorithms-7.5
  libocct-modeling-data-7.5 libocct-ocaf-7.5 libocct-visualization-7.5
```

carrotIndustries, 2023-04-03:
Instead of the conflicting packages, install `libocct-data-exchange-dev` and `libocct-ocaf-dev`

erusyd, 2023-04-03:
Thanks, succeeded.

closed 2023-04-04 by erusyd