horizon-eda/horizon #716: Board & Schematic only render background and grid

dgoodlad, 2023-04-04:
On my recent install of Horizon EDA, I can't get any of the board/schematic views to render anything but the background colour and the grid. The part and package viewers have the same behaviour. Nothing shows: no text, no wires (even when actively being drawn), no symbols, no footprints. 3d previews _do_ work.

[image]

I've got a Debian 12 virtual machine, using VMware Fusion's accelerated 3d graphics. From what I can tell this should be more than enough to support Horizon's needs:

```
glxinfo|grep version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.3
    Max compat profile version: 4.3
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL core profile version string: 4.3 (Core Profile) Mesa 22.3.6
OpenGL core profile shading language version string: 4.30
OpenGL version string: 4.3 (Compatibility Profile) Mesa 22.3.6
OpenGL shading language version string: 4.30
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 22.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
    GL_EXT_shader_implicit_conversions, GL_EXT_shader_integer_mix, 
```

I don't see any other graphical bugs or glitches. I've tested in both Xorg and in Wayland, same issue.

dgoodlad, 2023-04-04:
> Does changing the layer opacity or antialiasing settings do anything?

Disabling antialiasing fixed it! 1x, 2x, 4x sampling all make those layers disappear, while Off works fine.

dgoodlad, 2023-04-04:
I just tested a build with blending disabled by commenting out that line as you suggested, but it didn't have any effect. Enabling antialiasing still causes the same problem.

I'm not ruling out an issue with the VMware graphics driver here, or some odd interaction with my mac's graphics hardware. But any other ideas to troubleshoot?

carrotIndustries, 2023-04-04:
> But any other ideas to troubleshoot?

Another guess is that it's got to do something with geometry shaders since they're used for almost everything but the grid and cursor.

Can you post a screenshot of the edge of the board in the 3D view? The edges of the layers are the only thing the 3D view that uses geometry shaders.

dgoodlad, 2023-04-04:
[image]

carrotIndustries, 2023-04-05:
Sorry, my bad, this perspective makes it quite hard to tell if the layer walls are there. Switch to the "Front" view by clicking on the first box in the title bar. I'd expect the board to be hollow as on the right side of this screenshot: 
[image]

dgoodlad, 2023-04-05:
[image]