horizon-eda/horizon #723: Inaccurate text placement in symbol editor

LHSmicius, 2023-06-12:
This is the placement I made in Symbol editor
[image]

And this is how it looks in schematic (same order as in picture above)
[image]

Other example. Good old symbol looks good in schematic
[image]

but not in symbol editor
[image]

carrotIndustries, 2023-06-12:
Can you share the symbol and unit you made? You need to tweak the orientation and position of the text such that the text's origin is placed right above/below the symbol and the multi-line value is readable.

LHSmicius, 2023-06-13:
The symbol is in this PR https://github.com/horizon-eda/horizon-pool/pull/321.
I made it to look good in schematic, but now it looks weird in some orientations of symbol editor.
Maybe you are right, because rotating some text boxes helped me to make it look good in schematic. I will try to make it look good in both places and let you know if it's possible.

LHSmicius, 2023-06-14:
You were right - it is possible to make it look good in both places. It is necessary to mirror the text in some orientations.
Problem is that in "Symbol Editor" two rotations looks the same as mirror, but actually it's not the same. In "Text placement preview" window and in "Schematic Editor" this difference is visible and looks weird. Now I understand how to workaround with this issue, but I think it is possible to differently represent the text when it is mirrored in "Symbol Editor".