horizon-eda/horizon #727: 3rd_party/alphanum/alphanum.hpp:298:38: error: no template named 'binary_function' in namespace 'std'; did you mean '__binary_function'?

yurivict, 2023-07-10:
Build failure with clang-16 which now defaults to c++20.
```std::binary_function``` has been deprecated in c++17,  and you don't set the -std=c++xx argument, so the sources are treated as c++20.

Please define the C++ level in your makefiles, and remove deprecated functions.

Version: 2.5.0
FreeBSD 14

carrotIndustries, 2023-07-10:
> and you don't set the -std=c++xx argument

I do, so what's not working?

https://github.com/horizon-eda/horizon/blob/2a7aedcb2ae5ab5b478519452284034535ce8165/Makefile#L928

yurivict, 2023-07-10:
```std::binary_function``` is deprecated in c++17.
This means the C++ code is invalid: it uses a non-existent function.

carrotIndustries, 2023-07-10:
Here's the patch: https://github.com/horizon-eda/horizon/commit/1efad3028a1953af2cf62b518a54ffeb309977bc.patch

commit 1efad302 (2023-07-10): alphanum: remove deprecated std::binary_function

closes #727
closed 2023-07-10 by carrotIndustries