horizon-eda/horizon #736: Feature request: get-polygon, or, polygons as arguments to expand-polygon

jbayless, 2023-09-07:
Horizon's parameter programs have an aspect that makes them very hard to edit: polygon coordinate lists hard-coded in the program. The most common use case for the parameter program is to take some existing shape (such as a polygon on the copper layer) and expand or shrink it (to generate an offset polygon for the solder paste or solder mask layers). Why not express this workflow in the parameter program directly?

Here is an example of an existing parameter program, from the WDFN8 Pads 5-8 padstack:

```
get-parameter [ solder_mask_expansion ]
expand-polygon [ mask_polygon -320000 1185000 -500000 1185000 -500000 -1185000 -320000 -1185000 -320000 -1730000 250000 -1730000 250000 -1185000 1800000 -1185000 1800000 -765000 1330000 -765000 1330000 -535000 1800000 -535000 1800000 -115000 1330000 -115000 1330000 115000 1800000 115000 1800000 535000 1330000 535000 1330000 765000 1800000 765000 1800000 1185000 250000 1185000 250000 1730000 -320000 1730000 ]
0 get-parameter [ paste_mask_contraction ] -
expand-polygon [ paste_polygon_1 485000 0 485000 1185000 250000 1185000 250000 1730000 -320000 1730000 -320000 1185000 -500000 1185000 -500000 0 ]
0 get-parameter [ paste_mask_contraction ] -
expand-polygon [ paste_polygon_2 485000 0 1330000 0 1330000 115000 1800000 115000 1800000 535000 1330000 535000 1330000 765000 1800000 765000 1800000 1185000 485000 1185000 ]
0 get-parameter [ paste_mask_contraction ] -
expand-polygon [ paste_polygon_3 485000 0 485000 -1185000 250000 -1185000 250000 -1730000 -320000 -1730000 -320000 -1185000 -500000 -1185000 -500000 0 ]
0 get-parameter [ paste_mask_contraction ] -
expand-polygon [ paste_polygon_4 485000 0 1330000 0 1330000 -115000 1800000 -115000 1800000 -535000 1330000 -535000 1330000 -765000 1800000 -765000 1800000 -1185000 485000 -1185000 ]
```
Look at all those magic numbers... So hard to make any changes to this padstack!

What if all that could be replaced by something like this:

```
get-polygon-vertices [ copper_polygon ]
get-parameter [ solder_mask_expansion ]
expand-polygon-vertices [mask_polygon ]
0 get-polygon-vertices [ copper_polygon ]
get-parameter [ paste_mask_contraction ]
expand-polygon-vertices [paste_polygon ]
```

Or even this:

```
get-parameter [ solder_mask_expansion ]
expand-polygon-vertices [mask_polygon copper_polygon ]
get-parameter [ paste_mask_contraction ]
expand-polygon-vertices [paste_polygon copper_polygon ]
```

What do you think?

carrotIndustries, 2023-09-11:
Parameter programs should be idempotent, i.e. you should be able to run them any number of times on a single document with the output not changing. Expanding the polygon in-place would defeat that.

jbayless, 2023-09-12:
I see. Yes, it would require separately storing a polygon's "as-drawn" state from the shape of the polygon after the parameter program's execution, so that only the as-drawn state is usable as the polygon program input. Maybe that's too much complexity.