horizon-eda/horizon #744: Wrong usage of Board Editor CLI leads to SIGABRT instead of error message

alex-fu27, 2023-09-24:
I ran into this here: #743
The usage of the Board Editor is `horizon-imp -b <board file> <block file> <via directory>` and if one does not specify all of the files, this leads to:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
```
from `src/core/core_board.hpp:15`.

It would be nice to have an error message here that gives the correct usage.

alex-fu27, 2023-09-24:
Super weird, since I wrote this issue, it started telling me `wrong invocation` :confused:

carrotIndustries, 2023-09-24:
I'm happy to merge a PR for this, but this isn't a top priority for me since it's only used for debugging and development.