horizon-eda/horizon #745: Plane of a deleted Net will crash the Board Editor

alex-fu27, 2023-09-24:
To reproduce, create a Plane for a Net, then delete the Net:
1. Create a new project.
2. Open schematic.
3. Place two resistors. Connect them with a Net.
4. Open board, place resistors, route net.
5. Create a plane for the net.
6. Close Board Editor.
7. In schematic, delete the net. Now you have a plane referencing a net that does not exist.
8. Open board editor. Clicking the plane now crashes the editor.

Crashes can occur everywhere the `uuid_ptr` of the plane is dereferenced, e. g. [here](https://github.com/horizon-eda/horizon/blob/49973c6ed1fb68b48dffa9645e5aa515d8596af1/src/imp/imp_board.cpp#L1118) or [here](https://github.com/horizon-eda/horizon/blob/49973c6ed1fb68b48dffa9645e5aa515d8596af1/src/board/airwires.cpp#L241).

Workaround: For each plane in the `board.json`, get the Net UUID and grep for the UUID in the files. If there is none, delete the plane from the `board.json`.

closed 2023-09-24 by carrotIndustries

carrotIndustries, 2023-09-24:
Wondering why this went unnoticed for so long…

alex-fu27, 2023-09-25:
I guess normally you have planes of power supply nets, which never get deleted, but I have a plane of a random net, supposed to act as a heatsink. Thanks for the fix :slightly_smiling_face:

commit 70853236 (2023-09-24): board: properly handle missing nets and polygons for planes

changelog: Bugfixes/Board Editor: don't crash when opening a board with missing plane nets

fixes #745