horizon-eda/horizon #750: Rules check on fab output seems broken?

frmdstryr, 2023-10-04:
Well I think [#247](https://github.com/horizon-eda/horizon/issues/247) was working but something broke because it doesn't seem to be giving a warning/error anymore? If I open a project with errors and try to export it doesn't run rules. If I manually run rules and get errors it still lets me export without any warning.

Spent half the day trying to rework the qfn chips on my last board thinking I had a shorted solder joint. Finally gave up and pulled off all the chips to find it still shorted, then tested an unpopulated board and saw was that was also shorted...

Re-opened the board ran the rules and see I had moved vias and forgot to update planes.

[image]

frmdstryr, 2023-10-05:
It appears to be because it was a clearance check and the warning is not done for clearance rules.

carrotIndustries, 2023-10-05:
The clearance check isn't run on fab. output since it can take a very long time.

frmdstryr, 2023-10-05:
Hmm, yeah it is noticeably slower. I have 2 potential suggestions.

If there are any board changes since rules were last run show question asking to rerun? Not sure if this will help since it can easily be skipped (in my case I didn't think about the planes in the internal layers).

The other thought is to always do a post export run of checks with a status checkmark or something. This way it does not slow the export and is not in the way but can still alert if something fails (like CI). What do you think?