horizon-eda/horizon #751: Footprint generator generates incorrect pitch

LHSmicius, 2023-10-08:
I found that pitch calculation works incorrectly with SO, PSON and QFN packages (didn't test much yet) and it works good with SON package. Here are some examples.
SO:
[image]
[image]

PSON:
[image]
[image]

QFN:
[image]
[image]

SON works good (also tried pitch 0.65 and 1.27), but same problem with pitch 0.655 (rounds to 0.65 or 0.66):
[image]
[image]

Same issue mentioned in https://github.com/horizon-eda/horizon-pool/pull/334

frmdstryr, 2023-10-09:
Can you try changing round off to none?

LHSmicius, 2023-10-09:
> Can you try changing round off to none?
It helped. This trick solved the problem in all cases.

What actually this round-off feature suppose to do?

carrotIndustries, 2023-10-09:
> What actually this round-off feature suppose to do?

It says so in the tip at the bottom of the window:

> All features of the land pattern is rounded to this value, including pad sizes and placement.

IIRC it's an IPC thing.

LHSmicius, 2023-10-09:
Ok... That explains everything... Sorry for creating this nooby issue.

closed 2023-10-09 by LHSmicius