horizon-eda/horizon #763: Feature request: Import enclosure for 3d view / board design

frmdstryr, 2024-01-08:
I know it's already possible to export a board as a step file for using in other viewers, but... would it be hard to import an enclosure step model into horizon so I can design the board around that?

tkornack, 2024-01-08:
While you wait, you can just make a it component that's placed on the board at (0,0) and that's excluded from the BOM

carrotIndustries, 2024-01-08:
Now that I'm developing https://dune3d.org/ I think it makes more sense to add IDF support to both of them. This has the added bonus that it's a vendor-neutral format so theoretically works with any PCB/3D CAD.

frmdstryr, 2024-01-09:
> just make a it component that's placed on the board at (0,0) 

Good idea! That should work for the time being. 

Having proper support would still be nice.