horizon-eda/horizon #764: Horizon does not support OpenCascade 7.8.0

thierry-FreeBSD, 2024-02-26:
The build fails with the message:
```
===>   horizon-eda-2.5.0_3 depends on shared library: libTKSTEP.so - not found
*** Error code 1
```
There is no more libTKSTEP in OpenCascade 7.8.0.

bob-u, 2024-02-26:
From release notes (https://dev.opencascade.org/doc/overview/html/occt__upgrade.html):

Reorganized DE TK

DE TK components have been combined or separated based on specific CAD formats to support plug-in ability.

    Components now have a "TKDE" prefix. The available list includes TKDESTEP, TKDEOBJ, TKDEIGES, TKDEGLTF, TKDEVRML, TKDEPLY, TKDESTL.
    The DE DRAW TK has been updated in a similar way: DRAW components now have a "TKXSDRAW" prefix. The available list includes TKXSDRAWSTEP, TKXSDRAWOBJ, TKXSDRAWIGES, TKXSDRAWGLTF, TKXSDRAWVRML, TKXSDRAWPLY, TKXSDRAWSTL.

Migration problems may occur during configuration time or compile time. Ensure that you update your project configuration accordingly.

thierry-FreeBSD, 2024-02-27:
> None of the distros we have CI for already has opencascade 7.8 Any ideas how to reproduce this?

I guess that most distros won't upgrade OpenCascade as long as it breaks its consumers - at least this is what we are doing in FreeBSD: see https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=277294 .

Thus, ATM, you have to compile it yourself.

frmdstryr, 2024-03-10:
I'm using a locally built horizon with locally built OCCT 7.8.0 and it works fine here... didn't have to change anything.

```
$ lddtree .local/bin/horizon-imp
horizon-imp => .local/bin/horizon-imp (interpreter => /lib64/ld-linux-x86-64.so.2)
  ...
    libTKernel.so.7.8 => /home/user/.local/lib/libTKernel.so.7.8
    libTKMath.so.7.8 => /home/user/.local/lib/libTKMath.so.7.8
    libTKG3d.so.7.8 => /home/user/.local/lib/libTKG3d.so.7.8
        libTKG2d.so.7.8 => /home/user/.local/lib/libTKG2d.so.7.8
    libTKGeomBase.so.7.8 => /home/user/.local/lib/libTKGeomBase.so.7.8
    libTKBRep.so.7.8 => /home/user/.local/lib/libTKBRep.so.7.8
    libTKTopAlgo.so.7.8 => /home/user/.local/lib/libTKTopAlgo.so.7.8
        libTKGeomAlgo.so.7.8 => /home/user/.local/lib/libTKGeomAlgo.so.7.8
    libTKPrim.so.7.8 => /home/user/.local/lib/libTKPrim.so.7.8
    libTKBO.so.7.8 => /home/user/.local/lib/libTKBO.so.7.8
    libTKShHealing.so.7.8 => /home/user/.local/lib/libTKShHealing.so.7.8
    libTKHLR.so.7.8 => /home/user/.local/lib/libTKHLR.so.7.8
    libTKMesh.so.7.8 => /home/user/.local/lib/libTKMesh.so.7.8
    libTKCDF.so.7.8 => /home/user/.local/lib/libTKCDF.so.7.8
    libTKLCAF.so.7.8 => /home/user/.local/lib/libTKLCAF.so.7.8
    libTKXSBase.so.7.8 => /home/user/.local/lib/libTKXSBase.so.7.8
    libTKDESTEP.so.7.8 => /home/user/.local/lib/libTKDESTEP.so.7.8
        libTKDE.so.7.8 => /home/user/.local/lib/libTKDE.so.7.8
    libTKXCAF.so.7.8 => /home/user/.local/lib/libTKXCAF.so.7.8
        libTKVCAF.so.7.8 => /home/user/.local/lib/libTKVCAF.so.7.8
        libTKV3d.so.7.8 => /home/user/.local/lib/libTKV3d.so.7.8
        libTKService.so.7.8 => /home/user/.local/lib/libTKService.so.7.8
        libTKCAF.so.7.8 => /home/user/.local/lib/libTKCAF.so.7.8
...

```

frmdstryr, 2024-03-10:
If you patch out https://github.com/horizon-eda/horizon/blob/master/meson.build#L31-L35 does it work then?

thierry-FreeBSD, 2024-03-11:
I have just built it successfully with the attached patch
[horizon-2.5.0_Makefile.diff.txt](https://github.com/horizon-eda/horizon/files/14564227/horizon-2.5.0_Makefile.diff.txt).

But this is for the 2.5.0 release, and it does not contain a [meson.build](https://github.com/horizon-eda/horizon/blob/master/meson.build) file.

I'd like to provide a patch for meson.build according to the OpenCascade version, but I don't know how to test it with meson. Could you please explain?

frmdstryr, 2024-03-11:
Do you mean test the opencascade version or test building with meson? 

I haven't used meson much but this seems to work:
```meson
assert(opencascade.found(), 'OpenCASCADE or OCE is required')
if opencascade.version() >= '7.8.0'
    message('opencascade is >=7.8.0')
else
    message('opencascade <7.8.0')
endif
```

thierry-FreeBSD, 2024-03-11:
> I haven't used meson much but this seems to work:

Yes, that works, thanks!

aperezdc, 2024-09-14:
Arch Linux is [now shipping OpenCascade 7.8.x](https://archlinux.org/packages/extra/x86_64/opencascade/) (7.8.1 as of September 2024), applying the patch from PR https://github.com/horizon-eda/horizon/pull/767 worked for me.

carrotIndustries, 2024-09-14:
> Arch Linux is [now shipping OpenCascade 7.8.x](https://archlinux.org/packages/extra/x86_64/opencascade/) (7.8.1 as of September 2024), applying the patch from PR https://github.com/horizon-eda/horizon/pull/767 worked for me.

For me, it works without any changes at all after doing a clean build.

carrotIndustries, 2024-09-14:
There were some problems in the detection of the broken link args: afa25d7e50f75ced34047cf8c3ffd17e3090cffe fixes that

closed 2024-09-14 by carrotIndustries

aperezdc, 2024-09-18:
> There were some problems in the detection of the broken link args: [afa25d7](https://github.com/horizon-eda/horizon/commit/afa25d7e50f75ced34047cf8c3ffd17e3090cffe) fixes that
> 
> It's also working on freebsd: https://cirrus-ci.com/task/6393203128008704?logs=main#L68

This also worked for me, thanks for fixing!