horizon-eda/horizon #765: Feature: Select Via net from all net name list

LHSmicius, 2024-03-03:
Now "Place via" tool allows to select net from explicitly named nets only (Picture below. All net names are explicitly set)
[image]

Other tool "Edit Plane" allows to target any plane (including implicitly named as this one shown in picture below)
[image]

It would be very useful to have same functionality in "Place Via" net selection.

tkornack, 2024-03-03:
It would be better if you placed a via and then were asked which of the nets it penetrates it should connect to.

LHSmicius, 2024-03-03:
> It would be better if you placed a via and then were asked which of the nets it penetrates it should connect to. 

I like your idea. In most cases it should penetrate only one net, but if it shows more, it's a good indicator that maybe you placed via in the wrong place.

tkornack, 2024-03-03:
It’s quite common to have ground and power planes under every via. But I believe that the net you intend to connect is the plane on the active copper layer.