horizon-eda/horizon #770: Pool cache not getting invalidated

tkornack, 2024-04-06:
After pulling in changes to a pool on disk, and then opening a design that uses those those new/changed parts, it is often the case that the design will not display the parts correctly until I explicitly open the pool, which seems to cause it to update the pool cache. The opening the design works as expected.  

As a result, every time I pull in changes locally I always open up the pool, wait for the update, and close it before opening a design. I recall that Horizon now correctly invalidates the design's local cache when it's been updated, but I wonder if there's another bug interacting with the cache of a pool. I don't know enough to write articulately about it, but I can confirm that it is a consistent issue for my group.

carrotIndustries, 2024-04-19:
> After pulling in changes to a pool on disk, and then opening a design that uses those those new/changed parts,

Are you referring to the items cached in the project pool?

tkornack, 2024-04-20:
Apologies for the bad description. Here's an example I had just now. I had a project cloned locally that I had reviewed and closed. I pulled in new changes from a colleague (some capacitors were replaced with new ones, for example), and opened it to see the following errors:

[image]

Note that some of these new components and changes are absent. If I close the project completely and open it again, the new parts are now visible and no errors show. Is it a pool cache issue?

carrotIndustries, 2024-05-15:
> Is it a pool cache issue?

More or less, yes. What you're seeing is expected, though not ideal behavior. For the schematic editor to find the newly-pulled items they need to be in the pool's database. The database only gets updated automatically when opening a project.

tkornack, 2024-05-21:
I do believe the project was closed when the changes were pulled in. What does it look at to determine whether to rebuild the database?