horizon-eda/horizon #773: Orphaned plane pours under blind vias

tkornack, 2024-05-10:
A plane underneath a micro via still fills as though the via were connected. The rules check correctly identifies the fill as unconnected. 

To reproduce: create a three-layer board with two nets; ground and signal and place vias as shown. The middle and bottom layers have the signal layer
[image]

The blind via correctly connects to the middle layer
[image]

But the bottom layer is filled wrongly
[image]

carrotIndustries, 2024-05-10:
Does the linked branch fix it for you?

tkornack, 2024-05-11:
Yes!

commit e58fcdf7 (2024-05-15): plane update: correctly handle blind vias

looking at the junction's layer range is sufficient
considering the has_via member was only required and correct when there
were just through vias

changelog: Bugfixes/Board Editor: correctly handle unconnected fragments below blind/buried vias

closes #773
closed 2024-05-11 by tkornack