horizon-eda/horizon #778: The pop-up menu is slow or does not pop up

yueduz, 2024-06-14:
[录屏 2024-06-14 15-19-11.webm](https://github.com/horizon-eda/horizon/assets/1448159/59e28fc5-a6df-4f55-bb98-63f65c353062)
2.6.0 Luna installed from flathub

frmdstryr, 2024-06-18:
Can you launch it from a terminal and see if there is any error output?

yueduz, 2024-08-08:
There is no useful output when launching from the terminal. But I found a pattern: the menu does not pop up after clicking the menu button, but pops up immediately when the window loses focus.

katyo, 2024-08-09:
Oh, seemingly GTK still have such stucks... If I remember correctly I see same behavior about four or five years ago in some different programs.
Popups in horizon also stucks sometimes but it is not so critical for me.
Maybe such behavior related to handler functions execution duration or problems related to GTK itself.