horizon-eda/horizon #785: Segmentation fault when opening local pool

trembel, 2024-09-26:
When I try to open my local pool, I get a segmentation fault.
Version: horizon-eda 2.6.0-2, installed using AUR.

Backtrace (from gdb):
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
0x0000771a1ae26bb0 in ppoll () from /usr/lib/libc.so.6
(gdb) c
Continuing.
[New Thread 0x771a03f8f6c0 (LWP 54341)]
[Thread 0x771a03f8f6c0 (LWP 54341) exited]

Thread 1 "horizon-eda" received signal SIGSEGV, Segmentation fault.
0x0000771a1c5a4d98 in ?? () from /usr/lib/libgobject-2.0.so.0
(gdb) bt
#0  0x0000771a1c5a4d98 in ??? () at /usr/lib/libgobject-2.0.so.0
#1  0x0000771a1c595ca9 in ??? () at /usr/lib/libgobject-2.0.so.0
#2  0x0000771a1c595f32 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#3  0x0000771a1c595ff4 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#4  0x0000771a1bb54514 in ??? () at /usr/lib/libgtk-3.so.0
#5  0x0000771a1c583f6d in g_object_unref () at /usr/lib/libgobject-2.0.so.0
#6  0x0000771a1cb1f1a1 in Gtk::Object::_release_c_instance() ()
    at /usr/lib/libgtkmm-3.0.so.1
#7  0x0000771a1ca17a4c in Gtk::Box::~Box() () at /usr/lib/libgtkmm-3.0.so.1
#8  0x00006261678a1dd6 in ??? ()
#9  0x0000626167a98b89 in ??? ()
#10 0x0000626167a604e0 in ??? ()
#11 0x0000626167ac5af7 in ??? ()
#12 0x0000626167ac67a6 in ??? ()
#13 0x0000771a1ca74b70 in ??? () at /usr/lib/libgtkmm-3.0.so.1
#14 0x0000771a1c57482a in g_closure_invoke () at /usr/lib/libgobject-2.0.so.0
#15 0x0000771a1c5a58df in ??? () at /usr/lib/libgobject-2.0.so.0
#16 0x0000771a1c595ca9 in ??? () at /usr/lib/libgobject-2.0.so.0
#17 0x0000771a1c595f32 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#18 0x0000771a1c595ff4 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#19 0x0000771a1b9dd1f3 in ??? () at /usr/lib/libgtk-3.so.0
#20 0x0000771a1b87521c in ??? () at /usr/lib/libgtk-3.so.0
#21 0x0000771a1c595e1c in ??? () at /usr/lib/libgobject-2.0.so.0
#22 0x0000771a1c595f32 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#23 0x0000771a1c595ff4 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#24 0x0000771a1b998f24 in ??? () at /usr/lib/libgtk-3.so.0
#25 0x0000771a1c578f76 in g_cclosure_marshal_VOID__BOXEDv ()
    at /usr/lib/libgobject-2.0.so.0
#26 0x0000771a1c595e1c in ??? () at /usr/lib/libgobject-2.0.so.0
#27 0x0000771a1c595f32 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#28 0x0000771a1c595ff4 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#29 0x0000771a1b992e85 in ??? () at /usr/lib/libgtk-3.so.0
#30 0x0000771a1b99a813 in ??? () at /usr/lib/libgtk-3.so.0
#31 0x0000771a1b99b1ef in ??? () at /usr/lib/libgtk-3.so.0
#32 0x0000771a1b954386 in gtk_event_controller_handle_event ()
    at /usr/lib/libgtk-3.so.0
#33 0x0000771a1bb4ad52 in ??? () at /usr/lib/libgtk-3.so.0
--Type <RET> for more, q to quit, c to continue without paging--
#34 0x0000771a1caf7999 in Gtk::Widget::on_button_release_event(_GdkEventButton*) ()
    at /usr/lib/libgtkmm-3.0.so.1
#35 0x0000771a1caf073b in Gtk::Widget_Class::button_release_event_callback(_GtkWidget*, _GdkEventButton*) () at /usr/lib/libgtkmm-3.0.so.1
#36 0x0000771a1b8739aa in ??? () at /usr/lib/libgtk-3.so.0
#37 0x0000771a1c595e1c in ??? () at /usr/lib/libgobject-2.0.so.0
#38 0x0000771a1c595f32 in g_signal_emit_valist () at /usr/lib/libgobject-2.0.so.0
#39 0x0000771a1c595ff4 in g_signal_emit () at /usr/lib/libgobject-2.0.so.0
#40 0x0000771a1bb5a7cd in ??? () at /usr/lib/libgtk-3.so.0
#41 0x0000771a1b9e6aa5 in ??? () at /usr/lib/libgtk-3.so.0
#42 0x0000771a1b9e787b in gtk_main_do_event () at /usr/lib/libgtk-3.so.0
#43 0x0000771a1c73a457 in ??? () at /usr/lib/libgdk-3.so.0
#44 0x0000771a1c774e80 in ??? () at /usr/lib/libgdk-3.so.0
#45 0x0000771a1c620559 in ??? () at /usr/lib/libglib-2.0.so.0
#46 0x0000771a1c683157 in ??? () at /usr/lib/libglib-2.0.so.0
#47 0x0000771a1c61fa55 in g_main_context_iteration () at /usr/lib/libglib-2.0.so.0
#48 0x0000771a1c396c36 in g_application_run () at /usr/lib/libgio-2.0.so.0
#49 0x0000626167941bdf in ??? ()
#50 0x0000771a1ad40e08 in ??? () at /usr/lib/libc.so.6
#51 0x0000771a1ad40ecc in __libc_start_main () at /usr/lib/libc.so.6
#52 0x000062616796a205 in ??? ()
```

carrotIndustries, 2024-09-26:
Needs more info. Does this happen with other pools, also with a new empty pool? Did this start happening out of the blue?

trembel, 2024-10-21:
Somehow, this solved itself. I changed laptop recently, and now the conversion to the newest scheme was successful.

closed 2024-10-21 by trembel