horizon-eda/horizon #788: Fails to build with libgit2 1.8.2

nunotexbsd, 2024-10-20:
FreeBSD 14.1 amd64 / Meson 1.5.2 / clang 18.1.5

```
FAILED: horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o
c++ -Ihorizon-eda.p -I. -I.. -Isrc -I../src -I3rd_party -I../3rd_party -I../3rd_party/range-v3 -I/usr/local/include/gtkmm-3.0 -I/usr/local/lib/gtkmm-3.0/include -I/usr/local/include/atkmm-1.6 -I/usr/local/lib/atk
mm-1.6/include -I/usr/local/include/gtk-3.0/unix-print -I/usr/local/include/gdkmm-3.0 -I/usr/local/lib/gdkmm-3.0/include -I/usr/local/include/giomm-2.4 -I/usr/local/lib/giomm-2.4/include -I/usr/local/include/gtk-
3.0 -I/usr/local/include -I/usr/local/include/cairo -I/usr/local/include/at-spi2-atk/2.0 -I/usr/local/include/at-spi-2.0 -I/usr/local/include/atk-1.0 -I/usr/local/include/dbus-1.0 -I/usr/local/lib/dbus-1.0/includ
e -I/usr/local/include/libepoll-shim -I/usr/local/include/gio-unix-2.0 -I/usr/local/include/pangomm-1.4 -I/usr/local/lib/pangomm-1.4/include -I/usr/local/include/glibmm-2.4 -I/usr/local/lib/glibmm-2.4/include -I/
usr/local/include/cairomm-1.0 -I/usr/local/lib/cairomm-1.0/include -I/usr/local/include/sigc++-2.0 -I/usr/local/lib/sigc++-2.0/include -I/usr/local/include/pango-1.0 -I/usr/local/include/pixman-1 -I/usr/local/inc
lude/fribidi -I/usr/local/include/harfbuzz -I/usr/local/include/freetype2 -I/usr/local/include/libpng16 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/l
ocal/include/uuid -I/usr/local/include/pgm-5.2 -fdiagnostics-color=never -D_LIBCPP_HARDENING_MODE=_LIBCPP_HARDENING_MODE_FAST -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++17 -O2 -pipe -fstack-protector-stro
ng -isystem /usr/local/include -fno-strict-aliasing -isystem /usr/local/include -isystem /usr/local/include -DZMQ_BUILD_DRAFT_API=1 -D_THREAD_SAFE -pthread -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -DJSON_USE
_IMPLICIT_CONVERSIONS=0 -DINC_PODOFO_WITHOUT_DIRECTORY -MD -MQ horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o -MF horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o.d -o horizon-eda.p/src_pool
-prj-mgr_pool-mgr_pool_git_box.cpp.o -c ../src/pool-prj-mgr/pool-mgr/pool_git_box.cpp
../src/pool-prj-mgr/pool-mgr/pool_git_box.cpp:641:17: error: no matching function for call to 'git_commit_create'
  641 |             if (git_commit_create(&new_commit_oid, repo, "HEAD", signature, signature, NULL,
      |                 ^~~~~~~~~~~~~~~~~
/usr/local/include/git2/commit.h:359:17: note: candidate function not viable: no known conversion from 'value_type *' (aka 'git_commit **') to 'const git_commit **' for 10th argument
  359 | GIT_EXTERN(int) git_commit_create(
      |                 ^
  360 |         git_oid *id,
  361 |         git_repository *repo,
  362 |         const char *update_ref,
  363 |         const git_signature *author,
  364 |         const git_signature *committer,
  365 |         const char *message_encoding,
  366 |         const char *message,
  367 |         const git_tree *tree,
  368 |         size_t parent_count,
  369 |         const git_commit *parents[]);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[ 45% 374/802] c++
```

Maybe related to 1.8.2 backport:

> 
> v1.8.2
> 
> This release reverts a const-correctness change introduced in
> v1.8.0 for the git_commit_create functions. We now retain the
> const-behavior for the commits arguments from prior to v1.8.0.
> 
> This change was meant to resolve compatibility issues with bindings
> and downstream users.

But it builds fine with <= 1.8.1
[horizon-eda-2.6.0_2.log](https://github.com/user-attachments/files/17451748/horizon-eda-2.6.0_2.log)

Thanks

carrotIndustries, 2024-10-20:
Try the referenced commit and see if it fixes the build for you.

nunotexbsd, 2024-10-21:
@carrotIndustries

It ifxes build.
Thank you for very fast fix!

Cheers

commit 3b6b783d (2024-10-20): make it build with libgit 1.8.2

fixes #788
closed 2024-10-21 by nunotexbsd