horizon-eda/horizon #790: Compilation is broken: no matching function for call to 'git_commit_create'

yurivict, 2024-10-30:
```
../src/pool-prj-mgr/pool-mgr/pool_git_box.cpp:641:17: error: no matching function for call to 'git_commit_create'
  641 |             if (git_commit_create(&new_commit_oid, repo, "HEAD", signature, signature, NULL,
      |                 ^~~~~~~~~~~~~~~~~
/usr/local/include/git2/commit.h:359:17: note: candidate function not viable: no known conversion from 'value_type *' (aka 'const git_commit **') to 'git_commit *const *' for 10th argument
  359 | GIT_EXTERN(int) git_commit_create(
      |                 ^
  360 |         git_oid *id,
  361 |         git_repository *repo,
  362 |         const char *update_ref,
  363 |         const git_signature *author,
  364 |         const git_signature *committer,
  365 |         const char *message_encoding,
  366 |         const char *message,
  367 |         const git_tree *tree,
  368 |         size_t parent_count,
  369 |         git_commit * const parents[]);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

[log](https://pkg-status.freebsd.org/beefy18/data/main-amd64-default/p332c73b035e3_sf377a0c7dfa/logs/horizon-eda-2.6.0_3.log)

Version: 2.6.0
FreeBSD 14.1

carrotIndustries, 2024-10-30:
Apply these two commits as patches:
https://github.com/horizon-eda/horizon/commit/3b6b783db797b593119756821c04ac64b66b7ffa
https://github.com/horizon-eda/horizon/commit/d5a46d2d7e879e82d0d2798715c1369f8ae54fcd

nunotexbsd, 2024-10-31:
@carrotIndustries, I confirm that it fixes build.
@yurivict, I have commit ready to commit.
```diff
--- a/cad/horizon-eda/Makefile
+++ b/cad/horizon-eda/Makefile
@@ -6,6 +6,7 @@ CATEGORIES=     cad

 PATCH_SITES=   https://github.com/${GH_ACCOUNT}/${GH_PROJECT}/commit/
 PATCHFILES=    3b6b783db797b593119756821c04ac64b66b7ffa.patch:-p1 # Fix build with libgit2 1.8.2
+PATCHFILES+=   d5a46d2d7e879e82d0d2798715c1369f8ae54fcd.patch:-p1 # Fix build with libgit2 1.8.3

 MAINTAINER=    yuri@FreeBSD.org
 COMMENT=       EDA package for printed circuit board design
@@ -15,8 +16,6 @@ WWW=          https://horizon-eda.org/ \
 LICENSE=       GPLv3
 LICENSE_FILE=  ${WRKSRC}/LICENSE

-BROKEN_FreeBSD_15=     compilation error: no matching function for call to 'git_commit_create', see https://github.com/horizon-eda/horizon/issues/790
-
 BUILD_DEPENDS= ${LOCALBASE}/include/boost/optional.hpp:devel/boost-libs \
                cmake:devel/cmake-core \
                cppzmq>0:net/cppzmq \
diff --git a/cad/horizon-eda/distinfo b/cad/horizon-eda/distinfo
index c42d839493f0..9c414cc5beb8 100644
--- a/cad/horizon-eda/distinfo
+++ b/cad/horizon-eda/distinfo
@@ -1,5 +1,7 @@
-TIMESTAMP = 1729495536
+TIMESTAMP = 1730362431
 SHA256 (horizon-eda-horizon-v2.6.0_GH0.tar.gz) = e7e680a05b92ac8ab4b6a32fb8e3b17bc298245d3d3d9224e9b3f7fb55b81256
 SIZE (horizon-eda-horizon-v2.6.0_GH0.tar.gz) = 4358268
 SHA256 (3b6b783db797b593119756821c04ac64b66b7ffa.patch) = 7f168ab1af5e51b2e357a621f36ece6e55b690d7987ef343ac6ff1be37814573
 SIZE (3b6b783db797b593119756821c04ac64b66b7ffa.patch) = 2170
+SHA256 (d5a46d2d7e879e82d0d2798715c1369f8ae54fcd.patch) = 32327df19fc6f8b5c864dc785eaae7ac31b91d7ab8ae6a1384aebbafa5df411a
+SIZE (d5a46d2d7e879e82d0d2798715c1369f8ae54fcd.patch) = 4575
```

Thanks for quick fix.

yurivict, 2024-10-31:
I've committed it.

Thanks, Nuno!

closed 2024-10-31 by yurivict

nunotexbsd, 2024-10-31:
> Apply these two commits as patches: [3b6b783](https://github.com/horizon-eda/horizon/commit/3b6b783db797b593119756821c04ac64b66b7ffa) [d5a46d2](https://github.com/horizon-eda/horizon/commit/d5a46d2d7e879e82d0d2798715c1369f8ae54fcd)

libgit2 was updated to 1.8.4 hours ago and:

> 
> v1.8.4
> 
> We erroneously shipped v1.8.3 without actually including the change in v1.8.2. This release re-re-introduces the pre-v1.8.0 commit constness behavior.

It builds ok with 2 patches applied or just the first one which seems fine as it fixes build with 1.8.3 and 1.8.4.... but maybe upstream should check it better.

Thanks