horizon-eda/horizon #792: Drop geometry shaders

DemiMarie, 2024-12-14:
Geometry shaders are a very poor fit to hardware and have bad performance just about everywhere.  It’s better to use compute shaders instead.  Apple silicon and non-desktop GPUs don’t even implement them in hardware, requiring slow software emulation to have any support at all.

carrotIndustries, 2024-12-14:
I know, but so far I haven't found a good replacement. Take line rendering for example: At minimum a line needs start/stop coordinates and width. Right now, a geometry shader takes all of these parameters as a single vertex and emits a quad with some other parameters so that the fragment shader has everything it needs to take care of the round line caps.

Another approach I considered was using instanced rendering and using the line data as per-instance vertices. That however doesn't work as I need indexed rendering and there's no such thing as indexed instancing. Theoretically, one could side-step the built-in vertex fetch mechanisms by storing index and vertex data in SSBOs and reading them in the vertex shader. So far unfortunately I couldn't tell if that'll be enough an improvement over geometry shaders to be worth pursuing.

DemiMarie, 2024-12-14:
I think using compute shaders and skipping the vertex fetch is the recommendation, but I am not sure.  I just know that geometry shaders are considered inefficient and many GPUs don’t support them.