horizon-eda/horizon #793: How do I share parts between projects?

ii8, 2024-12-21:
Each time I create an entity, package or part, it automatically gets put in a project local pool and I am unable to save it anywhere outside of that pool's directory so I can't save it in the main pool. This means I'm having to go each time and outside of horizon-eda manually copy everything I make to the main pool for it to be usable from other projects. Is there some way to just use the main pool everywhere and not have project specific pools?

Can I just delete the pool directory in each project and symlink it to the main pool?

tkornack, 2024-12-23:
If you create parts in the local pool, you can move them to another pool by right-clicking on the part and selecting move to other pool.

ii8, 2024-12-23:
Neat, thanks.

closed 2024-12-23 by ii8