horizon-eda/horizon #796: The export paths should not be saved in the project files

oldrev, 2025-01-12:
Hello, 

I noticed that Horizon EDA saves the export path in the JSON file, which can cause issues when merging with Git. It is suggested to implement a feature that automatically converts the user provide path to a relative path within the project, or alternatively, do not save the export path in project-related files.

carrotIndustries, 2025-01-12:
If the path is within the project folder, it's automatically converted to a relative path. At least that's what happens for me.

oldrev, 2025-01-12:
Thanks for your quick reply.

On Windows, I pasted a path within my project into the "File Name" field, but the path conversion did not work, possibly due to the "\" character?

oldrev, 2025-01-12:
Oh, I see, it only works when the path is selected through the file dialog, and pasting the path directly is not effective.

carrotIndustries, 2025-01-12:
So everything's working as it should.

closed 2025-01-12 by carrotIndustries