horizon-eda/horizon #797: Feature Request: Support for Castellation PCB

oldrev, 2025-01-19:
Hello, I recently noticed that KiCAD has added support for creating castellation PCB, and I really hope Horizon can offer the same feature. 

I know that the edge-placed vias in Horizon can be recognized as castellated holes by the fab, but it would be great to have an option in the Padstack to show the castellated effect in the 3D preview. 

Horizon does need some more love, Thank you.

carrotIndustries, 2025-01-19:
So is it just a thing in the 3D preview? Started hacking something and it now looks like this: 

[image]

Obviously still need to make the pth barrel open at the board edge.

oldrev, 2025-01-19:
Great job! IMHO, the best approach would be to add a "Castellation" option to the "Type" dropdown in the Padstack Editor, so it won’t mess up with the DRC.

[image]

oldrev, 2025-01-25:
So, is the current 2.6 version safe enough to give it a try with castellated holes?

[image]

carrotIndustries, 2025-01-25:
Nice board!

I've made at least one board with castellated holes and it turned out as expected.

What ultimately matters are the gerbers and in there, having the hole cut by outline layer is the expected way of representing castellated holes. See https://jlcpcb.com/help/article/what-is-castellated-holes for example.

Also make sure to specify castellated holes when ordering your board since some PCB manufacturers charge extra for them.

oldrev, 2025-01-25:
Thanks for the reminder, I've done a few castellated boards with KiCad before. 

This board is just a fun project anyway，it's okay to fail. I'm not gonna spend $$$ on antenna tuning or FCC/CE certifications for it.

closed 2026-09-21 by oldrev