horizon-eda/horizon #801: Feature Request: Different opacity for zones and traces

trembel, 2025-02-24:
When working with traces and zones, it is hard to distinguish them, especially in multi-layer designs.
Different opacities for the two would help, similarly as it is done in Altium (photo from google):

[image]

carrotIndustries, 2025-02-25:
As much as I'd like to have this feature too, there's currently no easy way to implement this:

The renderer draws all triangles (planes get divided into triangles for rendering) for one layer in one pass. More complex pads or other items on the same layer also get turned into triangles, so there's no easy way of applying a different opacity to just planes.

oldrev, 2026-09-21:
It looks like we now have `TriangleInfo::Type::PLANE_FILL`, so there seems to be a chance we'll get to see this feature. 

I suggest keeping Pads/Vias/Tracks to 100% by default and copper pour planes to 50%.