horizon-eda/horizon #804: Update router

boarchuz, 2025-03-21:
Unfortunately there are performance and stability issues with the router version used in Horizon.

There's some discussion and videos here regarding this problem (now fixed):
(tldr- very slow, taxing, erratic router performance in walkaround mode)
https://gitlab.com/kicad/code/kicad/-/issues/16076
https://gitlab.com/kicad/code/kicad/-/issues/15607
https://forum.kicad.info/t/kicad-6-slow-unusable/35769/22

To give some perspective, even on a simple board with few collisions, any mouse movement can cause my PC with an i7-11800H and 64gb memory to grind to a halt for seconds at a time, with high CPU usage and fans ramping up. Making small adjustments on more complex layouts can be very tedious and time consuming, with the board editor occasionally crashing.

I've finally decided to roll back to 2.3.1 for now, where performance is great.

In researching this, I note that there have been improvements to the router upstream as recently as a couple months ago. It would be great to pull these into Horizon.

Thanks for Horizon, Lukas, it's brilliant.

carrotIndustries, 2025-03-21:
Interesting, for me this has never been that big of an issue. Can you share a project where you're seeing this behavior?

boarchuz, 2025-03-21:
I was able to reproduce this quickly with a very contrived project. I threw a few components onto a board, imported my board rules (significant?) and added traces and a polygon until finding the reproducible behaviour in the video. Note CPU usage skyrocketing whenever the router starts to get 'stuck'.

https://github.com/user-attachments/assets/2ccd5f21-3c19-4946-9572-790a6260c25d

[router_issue.zip](https://github.com/user-attachments/files/19389771/router_issue.zip)

Horizon v2.6.0-42-g65d99ecc
FWIW I opened the exact same project in a 2.3.1 build and didn't see any of this, the router was always very responsive no matter how or where I moved the mouse.

closed 2026-09-20 by carrotIndustries