horizon-eda/horizon #806: "std::exception: SEXPR is not a integer type!" when importing KiCAD symbol

auroraanna, 2025-04-27:
Upon importing the .kicad_sym from <https://files.seeedstudio.com/wiki/XIAO_WiFi/Resources/Seeeduino-XIAO-ESP32C3-KiCAD-Library.zip>, i got this prompt:
![a prompt from Horizon EDA. The title is "Exception in signal handler". The content reads "std::exception: SEXPR is not a integer type! Looks like this is a bug. Please report this and describe what you did previously."](https://github.com/user-attachments/assets/bae1b8dc-4706-4635-bf4e-99435b97a4a2)

Similar to #654.

carrotIndustries, 2025-04-27:
Which version did you try this with? I'm under the impression that this has already been fixed on latest `master`.

auroraanna, 2025-05-03:
i tried v2.6.0 but it works error free on master. should've checked, sorry.

closed 2025-05-03 by auroraanna