horizon-eda/horizon #807: Build error in pool-mgr/pool_git_box.cpp

ktrace, 2025-05-01:
```
FAILED: horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o 
x86_64-pc-linux-gnu-g++ -Ihorizon-eda.p -I. -I../horizon-2.6.0 -Isrc -I../horizon-2.6.0/src -I3rd_party -I../horizon-2.6.0/3rd_party -I../horizon-2.6.0/3rd_party/range-v3 -I/usr/include/gtkmm-3.0 -I/usr/lib64/gtkmm-3.0/include -I/usr/include/atkmm-1.6 -I/usr/lib64/atkmm-1.6/include -I/usr/include/gtk-3.0/unix-print -I/usr/include/gdkmm-3.0 -I/usr/lib64/gdkmm-3.0/include -I/usr/include/giomm-2.4 -I/usr/lib64/giomm-2.4/include -I/usr/include/gtk-3.0 -I/usr/include/cairo -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/atk-1.0 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/usr/include/elogind -I/usr/include/gio-unix-2.0 -I/usr/include/pangomm-1.4 -I/usr/lib64/pangomm-1.4/include -I/usr/include/glibmm-2.4 -I/usr/lib64/glibmm-2.4/include -I/usr/include/cairomm-1.0 -I/usr/lib64/cairomm-1.0/include -I/usr/include/sigc++-2.0 -I/usr/lib64/sigc++-2.0/include -I/usr/include/pango-1.0 -I/usr/include/pixman-1 -I/usr/include/fribidi -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/lib64/libffi/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/uuid -fdiagnostics-color=always -D_GLIBCXX_ASSERTIONS=1 -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=c++17 -O2 -march=x86-64 -pipe -DZMQ_BUILD_DRAFT_API=1 -pthread -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -DJSON_USE_IMPLICIT_CONVERSIONS=0 -DHAVE_SPNAV -MD -MQ horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o -MF horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o.d -o horizon-eda.p/src_pool-prj-mgr_pool-mgr_pool_git_box.cpp.o -c ../horizon-2.6.0/src/pool-prj-mgr/pool-mgr/pool_git_box.cpp
../horizon-2.6.0/src/pool-prj-mgr/pool-mgr/pool_git_box.cpp: In member function ‘void horizon::PoolGitBox::handle_pr()’:
../horizon-2.6.0/src/pool-prj-mgr/pool-mgr/pool_git_box.cpp:642:116: error: invalid conversion from ‘std::array<git_commit*, 2>::pointer’ {aka ‘git_commit**’} to ‘const git_commit**’ [-fpermissive]
  642 |                                   ("merge pr " + std::to_string(pr_id)).c_str(), tree, parents.size(), parents.data())
      |                                                                                                        ~~~~~~~~~~~~^~
      |                                                                                                                    |
      |                                                                                                                    std::array<git_commit*, 2>::pointer {aka git_commit**}
In file included from /usr/include/git2/repository.h:14,
                 from /usr/include/git2/annotated_commit.h:11,
                 from /usr/include/git2.h:11,
                 from ../horizon-2.6.0/src/pool-prj-mgr/pool-mgr/pool_git_box.hpp:9,
                 from ../horizon-2.6.0/src/pool-prj-mgr/pool-mgr/pool_git_box.cpp:1:

```

Build type: native build
Project name: horizon-eda
Project version: 2.6.0
C compiler for the host machine: x86_64-pc-linux-gnu-gcc (gcc 14.2.1 "x86_64-pc-linux-gnu-gcc (Gentoo 14.2.1_p20241221 p7) 14.2.1 20241221")
C linker for the host machine: x86_64-pc-linux-gnu-gcc ld.bfd 2.44
C++ compiler for the host machine: x86_64-pc-linux-gnu-g++ (gcc 14.2.1 "x86_64-pc-linux-gnu-g++ (Gentoo 14.2.1_p20241221 p7) 14.2.1 20241221")
C++ linker for the host machine: x86_64-pc-linux-gnu-g++ ld.bfd 2.44
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library pthread found: YES
Found pkg-config: YES (/usr/bin/x86_64-pc-linux-gnu-pkg-config) 2.4.3
Run-time dependency gtk+-3.0 found: YES 3.24.48
Run-time dependency gtkmm-3.0 found: YES 3.24.9
Run-time dependency glibmm-2.4 found: YES 2.66.7
Run-time dependency giomm-2.4 found: YES 2.66.7
Run-time dependency libpng found: YES 1.6.47
Run-time dependency epoxy found: YES 1.5.10
Run-time dependency sqlite3 found: YES 3.49.1
Run-time dependency libgit2 found: YES 1.8.4
Run-time dependency libzmq found: YES 4.3.5
Run-time dependency cppzmq found: YES 4.10.0
Run-time dependency libcurl found: YES 8.12.1
Run-time dependency librsvg-2.0 found: YES 2.58.5
Run-time dependency libarchive found: YES 3.7.9
Found CMake: /usr/bin/cmake (3.31.5)
Run-time dependency opencascade found: YES 7.8.1
Run-time dependency spnav found: YES 1.1
Run-time dependency python3 found: YES 3.12
Run-time dependency py3cairo found: YES 1.27.0
Run-time dependency osmesa found: NO (tried pkgconfig and cmake)
Run-time dependency glm found: YES 1.0.1
Has header "/usr/include/podofo-0.9/podofo/podofo.h" : NO 
Run-time dependency libpodofo09 found: NO (tried pkgconfig and cmake)
Run-time dependency libpodofo found: YES 0.10.3
Fetching value of define "__GLIBCXX__" : 20241221 
Message: using libstdc++
Library stdc++ found: YES
Library stdc++fs found: YES
Run-time dependency uuid found: YES 2.40.4
Program python3 found: YES (/usr/bin/python3)
Build-time dependency gio-2.0 found: YES 2.82.5
Program /usr/bin/glib-compile-resources found: YES (/usr/bin/glib-compile-resources)
Build targets in project: 23

carrotIndustries, 2025-05-01:
This should be fixed in d5a46d2d7e879e82d0d2798715c1369f8ae54fcd.

I should really try to get a release out the door soon.

carrotIndustries, 2025-06-09:
Release is done, closing.

closed 2025-06-09 by carrotIndustries