horizon-eda/horizon #808: Board editor: Airwires are incorrectly shadowed when Layer mode "Shadow other" is selected

cortices, 2025-05-30:
When i put the Board editor in Layer mode "Shadow other", i expect to see a grey background of the layers other than the one I'm working on, which I do.

However, this also applies to airwires, which makes it very difficult to distinguish them for routing while in this mode.

Currently, I switch between layer modes a lot to route a 4-layer board, because the screen is very busy in layer mode "As is".
Particularly with a dual-side-populated board, this makes it difficult to identify which component courtyards conflicting is ok, and which aren't, without using the "Shadow other" mode.

### Demonstration
Here is layer mode "As is", with visual business making it hard to see component outlines across layers, but airwires are visible.

[image]

Here is the same view in layer mode "Shadow other", showing that airwires have been coloured the same grey as background layers, making it very hard to spot them for routing.

[image]

carrotIndustries, 2025-05-30:
The right way to fix this is to take into account the layer range of airwires and the current work layer when rendering them.
 - In as-is mode, all airwires shown.
 - In shadow/work layer only mode, only airwires where the layer range includes the work layer are shown. Airwires from other layers are hidden.

The layer range of an airwire is determined by the union of its connection points. If an airwire connected to tracks on Inner 2 and Top, it will be shown if the work layer is between these two layers.

This could lead to some airwires being hidden unexpectedly if the shortest path isn't what's intended to be routed, but I think this is acceptable since one could just switch to as-is layer display mode.

carrotIndustries, 2025-05-30:
Well, this was surprisingly easy to fix, see the linked branch.

Not quite sure if i should put it in the upcoming release scheduled for this weekend or only merge it afterwards.
➕ One more feature for the release, there aren't that many after all
➖ Last-minute features always make me feel a bit uneasy…