horizon-eda/horizon #810: Power symbols pins on drawing lines do not produce a warning

fruchti, 2025-06-29:
Spurred by [this post](https://akko.erincandescent.net/notice/AvcRZ2fzuAkBBIIPNQ).

Using a graphic line instead of a net line produces a warning for regular symbols, but not for power symbols:
[image]

The GND symbol here is not actually connected, but it isn’t obvious if you don’t take a *really* close look.

tkornack, 2025-06-29:
The difference is that the ground is not on a junction; it's just on the line.

fruchti, 2025-06-30:
> The linked commit adds a warning for that.
> 
> This should prevent this kind of issue, i think?

Played around for a bit and it covers all the cases I could come up with, thanks! The only inconsistency I found was that drawing a rectangle or arc such that it connects with a net line (the net is first) does not produce a wire, but moving the rectangle or arc into place or drawing the net second does. There is still an X at the open junction of the net line, which makes the non-connected state clear though.