horizon-eda/horizon #814: Pull ECAD models from mouser for component footprint

ZeFoxii, 2025-08-11:
My idea is the ability to pull ECAD models from mouser and be able to import them into Horizon EDA to automatically populate the footprint of any component not already in Horizon.

oldrev, 2025-09-15:
That's a very good idea, but have you ever heard of something called copyright?

atoav, 2026-09-10:
As @oldrev already mentioned copyright is the issue why this has not been done before in a way where parts are added to the pool. If we would do that, this would technically constitute redistribution. I checked on mouser and it is unclear under which license they distribute those files especially the 3D models, which may come from a heap of different manufacturers.

What is legally more okay is to use scripts to create parts from manufacturer tables and datasheets and create your own stuff from that.